robotpy/robotpy-cscore #7: Error building bdist_wheel on OSX

virtuald, 2017-01-21:
Must be something in pybind11's setup script.

joshsizer, 2019-01-12:
I was wondering if you might be able to help me install on Mac OSX 10.14.2, or atleast tell me that it's not possible at the moment.

Installed on system:
clang: Apple LLVM version 10.0.0 (clang-1000.10.44.4)
opencv: 4.0.1 and 2.4.13.7_2
python: 3.7
numpy: 1.15.4

I get slightly different errors when I use opencv 4 vs opencv 2. It always fails compiling ndarray_converter.cpp.

Here is the error log with opencv4 and opencv2 respectively: 
[install_fail_opencv4.log](https://github.com/robotpy/robotpy-cscore/files/2752271/install_fail_opencv4.log)
[install_fail_opencv2.log](https://github.com/robotpy/robotpy-cscore/files/2752290/install_fail_opencv2.log)

Any help would be appreciated!

auscompgeek, 2019-01-12:
OpenCV 4 is not supported: #39. OpenCV 2 only supports Python 2.

robotpy-cscore is only compatible with OpenCV 3.

joshsizer, 2019-01-12:
Ah, this was a suspicion of mine, but there is no opencv3 package on Homebrew. I am starting to think I'll have to compile from source. Do you have any resource that steps me through that process?

joshsizer, 2019-01-12:
Thanks for the quick responses. This project is awesome, I'd like to say thank you for all the hard work.

auscompgeek, 2019-01-30:
FYI it seems the Homebrew folk have come to their senses and have an `opencv@3` package now.

auscompgeek, 2019-01-30:
Looks like the source package on PyPI actually builds now.

closed 2019-01-30 by auscompgeek