robotpy/robotpy-cscore #11: NumPy error on RoboRIO

czhao39, 2017-02-07:
I am able to deploy my camera code successfully (from Linux), but on the RoboRIO, I get the following error:

>File "/home/lvuser/py/camera.py", line 4, in \<module\> 
     import cv2 
 ImportError: numpy.core.multiarray failed to import 

Searching through Google, it seems that the source of this error is usually having multiple versions of NumPy installed. However, I ran `find -name numpy` from the root of the RoboRIO and I only have NumPy 1.12.0 installed. I installed it by running `robotpy-installer download-opkg python36-numpy`.

virtuald, 2017-02-07:
Hm. I haven't tried it on a fresh RoboRIO yet. That bug really should be filed over at roborio-packages too. Let me try it real quick on my VM.

virtuald, 2017-02-07:
Works for me. What version of OpenCV did you install?

czhao39, 2017-02-07:
That's odd. I used the installer (`robotpy-installer install-opkg opencv3-dev` and `robotpy-installer install-opkg python36-opencv3`).

czhao39, 2017-02-07:
If it worked for you, maybe I'll just try reinstalling everything tomorrow. Is there a convenient way to reinstall opkg packages installed with robotpy-installer?

virtuald, 2017-02-07:
I assume you have the 2017 RoboRIO image installed? If you haven't updated, that could cause issues.

* You might want to try rebooting the RoboRIO. 
* You might want to try ssh'ing in as admin (no password required), and run 'ldconfig'. Shouldn't be necessary but I've had issues with that in the past...
* You can ssh in as admin (no password required), and there's a directory called 'opkg_cache'. You can do `opkg remove PACKAGENAME`, and then do `opkg install FILENAME` with the files in that directory.

Let me know how it goes.

czhao39, 2017-02-07:
Thanks! I do have the 2017 image installed. I'll let you know how it goes tomorrow.

czhao39, 2017-02-08:
Well, the error magically went way after a restart (I should've thought of that). But now I get the following error:

>camera = cs.UsbCamera("usbcam", port) 
AttributeError: module 'cscore' has no attribute 'UsbCamera'

Since the RoboRIO is running Linux, will the `UsbCamera` class eventually be supported, or will we have to use OpenCV instead as you described in the readme? I got cscore with `robotpy-installer install-opkg python36-robotpy-cscore`.

virtuald, 2017-02-08:
cscore is still a preview release (any day now!), so there are still bugs. I expect UsbCamera to work on the RIO any day now.

czhao39, 2017-02-08:
Okay, thanks! Just to let you know, it seems like `cscore` doesn't actually have any classes, not just `UsbCamera`. I ran `help(cs)` and got the following output (no package contents):

>Help on package cscore: 
>  
> NAME 
>     cscore 
>  
> PACKAGE CONTENTS 
>  
>  
> FILE 
>     /home/lvuser/py/cscore/__init__.py

closed 2017-02-08 by czhao39

auscompgeek, 2017-02-08:
@czhao39 It looks like you've somehow have an empty `cscore/__init__.py` in your code. That'll override the real cscore, which should be installed in `/usr/local/lib/python3.6/site-packages/cscore`, I believe.

virtuald, 2017-02-08:
@czhao39 did you compile it yourself or install the ipk I uploaded? Maybe it's broken. Either way, I'm trying to get it done soon.

czhao39, 2017-02-08:
I'm sorry, @auscompgeek was right. I just realized I completely forgot to delete the dummy `cscore` module I made in Windows for testing, which I didn't have in Linux. That was really dumb. I'll try again on Thursday and see if the NumPy error persists. Also, thanks for everything you've done / are doing!