robotpy/robotpy-cscore #17: Error on Windows 10, please help!

Leo428, 2017-03-23:
The pip installation went well until: 
`
src/_cscore.cpp(12): fatal error C1083: Cannot open include file: 'opencv2/opencv.hpp': No such file or directory
  error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.10.25017\\bin\\HostX86\\x86\\cl.exe' failed with exit status 2
  ----------------------------------------
  Failed building wheel for robotpy-cscore
`
### Is there any solution for this issue or which system is currently supported? Thank you!

virtuald, 2017-03-23:
I've never tried building it on Windows. You would need to help it find your OpenCV installation for it to work. There are probably arguments to pip you can use?

Leo428, 2017-03-23:
okay, let me try to reinstall the opencv stuff ...

virtuald, 2017-03-23:
How do you have OpenCV installed?

virtuald, 2017-03-23:
Also, you may find it easier to get realtime help at https://gitter.im/robotpy/robotpy-wpilib if you're doing it now.

Leo428, 2017-03-23:
I installed opencv by downloading the opencv for windows and extracted it. I tried to add the path into system variable. In python 3.6, I am able to import cv2.

virtuald, 2017-03-23:
Yeah, the thing you would need to do then is put the include/lib files somewhere that python can find them. Maybe in the 'include' directory for site-packages? 

Another route you could do is add an environment variable to setup.py, and tell it where the OpenCV install is.

virtuald, 2017-03-23:
You can also figure out where python keeps its stuff by running `py -3 -c "import sysconfig; print(sysconfig.get_path('include'))"` ... there's a bunch of variables you can look at: https://docs.python.org/3/library/sysconfig.html

Leo428, 2017-03-23:
### Hi, my dear friend! 
### Now the cannot find path has been solved by compiling opencv from source. However, another and the only issue popped up! Please Help! 
`
     C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\users\huzhe\appdata\local\programs\python\python36\Include -IC:\Users\huzhe\AppData\Roaming\Python\Python36\Include -Icscore_src/include -Intcore_src/wpiutil/include -Ic:\users\huzhe\appdata\local\programs\python\python36\lib\site-packages\numpy\core\include -Ic:\users\huzhe\appdata\local\programs\python\python36\include -Ic:\users\huzhe\appdata\local\programs\python\python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\winrt" /EHsc /Tpsrc/_cscore.cpp /Fobuild\temp.win-amd64-3.6\Release\src/_cscore.obj /EHsc /DVERSION_INFO=\"2017.0.2\"
    _cscore.cpp
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(6): error C2065: 'f': undeclared identifier
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(6): error C2275: 'Args': illegal use of this type as an expression
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(5): note: see declaration of 'Args'
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(6): error C2988: unrecognizable template declaration/definition
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(6): error C2059: syntax error: '...'
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(6): error C2143: syntax error: missing ';' before '{'
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(6): error C2447: '{': missing function header (old-style formal list?)
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(15): error C2988: unrecognizable template declaration/definition
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(15): error C2059: syntax error: '<tag>::*'
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(15): error C2143: syntax error: missing ';' before '{'
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(15): error C2447: '{': missing function header (old-style formal list?)
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(24): error C2988: unrecognizable template declaration/definition
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(24): error C2059: syntax error: '<tag>::*'
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(24): error C2143: syntax error: missing ';' before '{'
    c:\users\huzhe\appdata\local\temp\pip-build-bed506s4\robotpy-cscore\src\release_gil.hpp(24): error C2447: '{': missing function header (old-style formal list?)
    ntcore_src/wpiutil/include\llvm/Hashing.h(79): error C2059: syntax error: ':'
    ntcore_src/wpiutil/include\llvm/Hashing.h(93): fatal error C1903: unable to recover from previous error(s); stopping compilation
    Internal Compiler Error in C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\HostX86\x86\cl.exe.  You will be prompted to send an error report to Microsoft later.
    INTERNAL COMPILER ERROR in 'C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\HostX86\x86\cl.exe'
        Please choose the Technical Support command on the Visual C++
        Help menu, or open the Technical Support help file for more information
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.10.25017\\bin\\HostX86\\x86\\cl.exe' failed with exit status 2
`

### Please help! I really need this to work. Crying! And What System do you use when you use the cscore lib? Thank you!

virtuald, 2017-03-23:
I believe that's an error related to pybind11 and Visual Studio 2017. It looks like they just released a new version of pybind11 yesterday, I haven't tested with it yet. I also don't have VS 2017 (or Windows with a compiler easily available), so I can't test that either.

What I would recommend to you is separate the cscore portion of your code from the opencv portion, that way you can run the code and test your image processing with static images. maybe next year we'll have a better way to do that. But basically, only import cscore when you're actually using it, not at the toplevel of your module.

Hopefully next year we'll get Windows support working, but it sounds like it isn't going to work for you. Anyways, it's not super useful on Windows, as USBCamera isn't supported there yet.

auscompgeek, 2019-01-12:
Ok, I spent way too long trying to get this all to compile on Windows. Thankfully with a bit of help and some massaging of the setup.py, I managed to build this. Here's how I did it:

1. Install conda.
2. Create a conda env with opencv installed, and enter it.
3. Set the following environment variables:
```batch
set CL=/I%CONDA_PREFIX%\Library\include /std:c++14 /DNOMINMAX
set LIB=%CONDA_PREFIX%\Library\lib
```
4. Edit `setup.py` so the opencv library names are versioned (e.g. append 342 to each of them - look in `%LIB%` for the version number).
5. `python setup.py bdist_wheel`

Here's the wheel I built for Python 3.7 on x64 with OpenCV 3.4.2: https://vovo.id.au/tmp/robotpy_cscore-2019.0.2.post0.dev0-cp37-cp37m-win_amd64.whl

auscompgeek, 2020-02-09:
I think with our pkg-config support now and the Windows fixes I put in last year, building with conda-forge should be quite doable without a lot of effort. I think we're forgetting to pass the `/std:c++17` flag though.

If we can make compilation "just" work, we should submit to conda-forge.

auscompgeek, 2021-01-02:
After attempting to build on Windows with conda-forge, it turns out that OpenCV does not generate pkg-config info at all on Windows, see https://github.com/conda-forge/opencv-feedstock/pull/254