robotpy/robotpy-cscore #18: Report Bug during installing on Ubuntu

Leo428, 2017-03-25:
`
    sudo pip install robotpy-cscore 
`
### Error:
`
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/local/include/python3.6 -I/home/zheyuan/.local/include/python3.6m -Icscore_src/include -Intcore_src/wpiutil/include -I/usr/local/lib/python3.6/dist-packages/numpy/core/include -I/usr/include/python3.6m -c src/_cscore.cpp -o build/temp.linux-x86_64-3.6/src/_cscore.o -DVERSION_INFO="2017.0.2" -s -g0 -std=c++14 -fvisibility=hidden
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    In file included from src/_cscore.cpp:6:0:
    src/release_gil.hpp:6:70: error: expected ‘,’ or ‘...’ before ‘PYBIND11_NOEXCEPT_SPECIFIER’
         std::function<Return(Args...)> release_gil(Return (*f)(Args ...) PYBIND11_NOEXCEPT_SPECIFIER) {
                                                                          ^
    src/release_gil.hpp:15:85: error: expected ‘,’ or ‘...’ before ‘PYBIND11_NOEXCEPT_SPECIFIER’
         std::function<Return(Class*, Args...)> release_gil(Return (Class::*f)(Args ...) PYBIND11_NOEXCEPT_SPECIFIER) {
                                                                                         ^
    src/release_gil.hpp:24:91: error: expected ‘,’ or ‘...’ before ‘PYBIND11_NOEXCEPT_SPECIFIER’
         std::function<Return(Class*, Args...)> release_gil(Return (Class::*f)(Args ...) const PYBIND11_NOEXCEPT_SPECIFIER) {
                                                                                               ^
    src/_cscore.cpp: In function ‘PyObject* pybind11_init()’:
    src/_cscore.cpp:32:6: warning: ‘pybind11::capsule::capsule(const void*, void (*)(PyObject*))’ is deprecated: Please pass a destructor that takes a void pointer as input [-Wdeprecated-declarations]
         });
          ^
    In file included from /usr/local/include/python3.6/pybind11/cast.h:13:0,
                     from /usr/local/include/python3.6/pybind11/attr.h:13,
                     from /usr/local/include/python3.6/pybind11/pybind11.h:36,
                     from src/_cscore.cpp:2:
    /usr/local/include/python3.6/pybind11/pytypes.h:1015:5: note: declared here
         capsule(const void *value, void (*destruct)(PyObject *))
         ^
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
`
### I think the new pybind11 has some issue in it.

Leo428, 2017-03-25:
`
pip3 install 'pybind11==2.0.1'
`
### After I went back to this version, cscore installed fine. So the new version definitely has bug in it.

virtuald, 2017-03-26:
Yeah, pybind11 just released a new version. I'll try to address that next week, thanks for reporting the bug!

virtuald, 2017-09-17:
I never got around to doing this, but it's on the agenda for 2018.

RustyRaptor, 2017-11-02:
Thanks, this fixed it.

commit 156adb10 (2017-12-11): Update pybind11

- Fixes #18
closed 2017-12-11 by virtuald