robotpy/robotpy-cscore #22: Fix compilation on Jetson

virtuald, 2017-09-23:
I'm pretty sure OpenCV on the Jetson is OpenCV v2, and we don't support that in cscore. We should see how difficult that would be, because there are teams that would use it if it were available.

sparks58, 2017-12-12:
Yes, that is correct. The Jetson comes installed with opencv 2.4. Cscore supports version 3, so you'll need to either install version 3 or compile it yourself. Here is a script that'll compile OpenCV for you: http://www.jetsonhacks.com/2017/04/05/build-opencv-nvidia-jetson-tx2/

virtuald, 2017-12-12:
I believe the primary issue we have with compiling on a Jetson is that our ndarray <-> mat conversion routines only support OpenCV 3. 

However, you're correct, if someone were to install OpenCV 3.x on a Jetson, this package would work fine.

virtuald, 2018-12-30:
We're not supporting Python 2 in pynetworktables, and I don't forsee it happening for cscore either unfortunately.

closed 2018-12-30 by virtuald