robotpy/robotpy-cscore #28: Module Upgrade Fails on Jetson TX1

FRC2539, 2018-01-18:
We are trying to upgrade cscore on our Jetson TX1. We installed the earlier version during the summer. Now when we run `sudo pip3 install --upgrade robotpy-cscore` the process fails with:

```
aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/local/include/python3.5 -I/home/ubuntu/.local/include/python3.5m -Icscore_src/src/main/native/include -Iwpiutil_src/src/main/native/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/include/python3.5m -c src/_cscore.cpp -o build/temp.linux-aarch64-3.5/src/_cscore.o -DVERSION_INFO="2018.0.0" -s -g0 -std=c++14 -fvisibility=hidden
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
src/_cscore.cpp:18:13: error: ‘call_guard’ in namespace ‘py’ does not name a template type
     typedef py::call_guard<py::gil_scoped_release> release_gil;
                 ^
src/_cscore.cpp:21:16: error: expected constructor, destructor, or type conversion before ‘(’ token
     PYBIND11_MODULE(_cscore, m) {
                    ^
error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
```

We have OpenCV 3.4 installed and working in Python 3.

virtuald, 2018-01-18:
You need pybind11 >= 2.2 installed.

FRC2539, 2018-01-18:
Thank you. Is it possible to make that a requirement of the robotpy-cscore package?

auscompgeek, 2018-01-18:
It… sort of is. There's a `setup_requires` declared in the setup.py file. It doesn't seem to get picked up by pip though. It's not a runtime requirement of robotpy-cscore, so it doesn't make sense to make it an `install_requires` (otherwise it'd have to remain installed to use robotpy-cscore).

FRC2539, 2018-01-18:
Okay. We'll add it to our project's requirements.txt, since we don't care if it remains installed.

closed 2018-01-18 by FRC2539