robotpy/robotpy-cscore #58: Another non-unicode issue

virtuald, 2019-01-20:
We fixed a problem similar to this previously. 

https://www.chiefdelphi.com/t/trouble-running-the-python-example-code-with-wpilib-frcvision-raspberry-pi-image/342147/2

Would be good to identify the string in question and figure out the right way to convert it to unicode.

virtuald, 2019-01-20:
I'm not sure what happens if we store bytes in pynetworktables (even though this is clearly what the C ntcore implementation must be doing). I think the python version will try to convert it to a string, which will fail.

virtuald, 2019-01-20:
If anyone has a camera that causes this issue, to address this we first need to know what the bytes returned by the camera are. The best way to do this is to (a) identify what key is causing the issue (typically it's the camera name) and (b) get the bytes (using... some program? A good start would be the output of `v4l2-ctl --all` or some similar program).

PeterJohnson, 2019-01-20:
The problem is the intmenu's.  cscore does the wrong thing when converting the enum value names.  I opened a PR to fix this in wpilibsuite/allwpilib#1561.

closed 2019-02-04 by auscompgeek