robotpy/robotpy-cscore #60: segfault in in cs::MjpegServerImpl::ConnThread::SendStream

ariovistus, 2019-01-20:
robotpy-cscore==2019.0.3, fedora 28 with their packaged opencv, seems to be 3.4.1

I run the following code:
```
import time
from cscore import CameraServer, VideoSource, UsbCamera
inst = CameraServer.getInstance()
camera = UsbCamera()
server = inst.startAutomaticCapture(camera=camera, return_server=True)
camera.setConnectionStrategy(VideoSource.ConnectionStrategy.kKeepOpen)

while True:
    time.sleep(10)
```

as my first attempt to do something with cscore. When I try to connect to http://localhost:1181, the segfault occurs. gdb doesn't give me much:
```
0x00007fffea0120a2 in cs::MjpegServerImpl::ConnThread::SendStream(wpi::raw_socket_ostream&) ()
   from /home/ellery/.virtualenvs/tabularasa/lib/python3.6/site-packages/cscore/_cscore.cpython-36m-x86_64-linux-gnu.so
```
workaround is simple:
```
camera = UsbCamera("name", 0)
```
and everything works, but `help(UsbCamera)` suggests the former invocation is valid:
```
 |  __init__(...)
 |      __init__(*args, **kwargs)
 |      Overloaded function.
 |      
 |      1. __init__(self: cscore._cscore.UsbCamera) -> None
 |      
 |      2. __init__(self: cscore._cscore.UsbCamera, name: wpi::Twine, dev: int) -> None

```

auscompgeek, 2019-01-21:
A bunch of the VideoSink and VideoSource subclasses appear to have default no-arg constructors. @PeterJohnson are these intended to be usable?

PeterJohnson, 2019-02-03:
The no-args constructor initializes the C++ wrapper object with an invalid handle (handle=0).  No underlying implementation object is created in this case, and calling other functions on that object won't work (they will return an error / set the error flag in the class).  That should simply result in nothing happening, not in a crash though.

PeterJohnson, 2019-02-03:
It looks like I may be missing a couple of null checks in MjpegServerImpl.  Will fix.

auscompgeek, 2019-02-03:
I suppose we should also not expose the no-args constructors as well, since they're not very useful from Python land.