robotpy/robotpy-cscore #62: error installing cscore bc of numpy

pchsRoboticPgram, 2019-01-23:
Trying to install robotpy-cscore, but program is not finding numpy even though I have it installed. I am using pycharm. pls help us. Also is there another way besides this method to access the axis camera or USB camera on to the smart dashboard?
[image]
[image]
[image]

ArchdukeTim, 2019-01-23:
Looks like it's not installing robotpy-cscore in the same venv as your project

pchsRoboticPgram, 2019-01-23:
how can this be fixed? how do is it try to install on a different environment? Sorry I'm new to this kind of stuff.

ArchdukeTim, 2019-01-23:
I think if you run `C:\Desktop\KyleAndDaitensFirstProgram\venv\Scripts\python.exe -m pip install robotpy-cscore` it should work

pchsRoboticPgram, 2019-01-23:
[image]

OK this came up...

auscompgeek, 2019-01-23:
Note that robotpy-cscore currently does not support installation on Windows due to a couple of issues, see #59.

auscompgeek, 2019-01-23:
Seems likely that doing `import numpy` in C:\Desktop\KyleAndDaitensFirstProgram\venv\Scripts\python.exe would equally break. If so, your virtualenv is broken.

pchsRoboticPgram, 2019-01-23:
is there a way to make the axis camera show up on the smart dashboard without using the CameraServer class from robotpy-cscore?

auscompgeek, 2019-01-23:
It's possible, but not as fun. You would have to add the stream values to NetworkTables yourself. You can take a read of the CameraServer class to figure out how to do that.

Note that you don't need to install robotpy-cscore on your dev machine to use it on the robot. The `wpilib.CameraServer` shim will only launch the cscore code if it's on the robot.

pchsRoboticPgram, 2019-01-24:
Thank you so much! We installed robotpy-cscore directly to the RoboRIO via command prompt and it worked like a charm!

closed 2019-01-24 by pchsRoboticPgram