robotpy/robotpy-cscore #67: Error while installing on Raspberry PI

PepperProgramming, 2019-02-13:
Sorry for duplicate bug report - I just noticed the other one was closed and so opening a new issue.

Hi, I am trying to install on RaspberryPi and I am getting following error.
I tried pip3 install robotpy-cscore==2019.0.0a01 and it failed after few hours

ython3.5m -c src/_cscore.cpp -o build/temp.linux-armv7l-3.5/src/_cscore.o -DVERSION_INFO="2019.0.0a1" -s -g0 -std=c++14 -fvisibility=hidden -D_GNU_SOURCE
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
arm-linux-gnueabihf-gcc: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See file:///usr/share/doc/gcc-6/README.Bugs for instructions.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 4

Failed building wheel for robotpy-cscore
Running setup.py clean for robotpy-cscore
Failed to build robotpy-cscore
Could not build wheels for {} which use PEP 517 and cannot be installed directly
(cv) pi@OpenCV:~ $

virtuald, 2019-02-13:
Hm. My guess is that you ran out of memory while compiling. How much memory does your Pi have?

PepperProgramming, 2019-02-13:
Thanks for the pointer, that helped. 

For others who are trying to install on RPi, here is what I did
$ sudo nano /etc/dphys-swapfile

# set size to absolute value, leaving empty (default) then uses computed value
#   you most likely don't want this, unless you have an special disk situation
# CONF_SWAPSIZE=100
CONF_SWAPSIZE=2048

$ sudo /etc/init.d/dphys-swapfile stop
$ sudo /etc/init.d/dphys-swapfile start

pip3 install robotpy-cscore==2019.0.0a01

after installation is successful, revert swapsize to original to avoid burning your SD card.

virtuald, 2019-02-13:
@PepperProgramming glad to hear that it fixed your issue!

One thing I noticed -- recommend that you use the current version of robotpy-cscore, and not 2019.0.0a1, as there are useful bugfixes in the latest releases.

PepperProgramming, 2019-02-13:
you just run 
pip3 install robotpy-cscore

PepperProgramming, 2019-02-13:
When I tried pip3 install robotpy-cscore, installation failed again. I increased my swapfile size and still no luck. Here is the o/p.

(cv) pi@OpenCV:~ $ pip3 uninstall robotpy-cscore==2019.0.0a1
Uninstalling robotpy-cscore-2019.0.0a1:
  Would remove:
    /home/pi/.virtualenvs/cv/lib/python3.5/site-packages/cscore/*
    /home/pi/.virtualenvs/cv/lib/python3.5/site-packages/robotpy_cscore-2019.0.0a1.dist-info/*
Proceed (y/n)? y
  Successfully uninstalled robotpy-cscore-2019.0.0a1
(cv) pi@OpenCV:~ $ pip3 install robotpy-cscore
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting robotpy-cscore
  Using cached https://files.pythonhosted.org/packages/15/dd/85cab4ad0e71c0c3e8b3d8a4d67cbd87529ad67a93a7e135758f9ad74017/robotpy-cscore-2019.0.7.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: pynetworktables in ./.virtualenvs/cv/lib/python3.5/site-packages (from robotpy-cscore) (2019.0.0)
Requirement already satisfied: numpy in ./.virtualenvs/cv/lib/python3.5/site-packages (from robotpy-cscore) (1.16.0)
Building wheels for collected packages: robotpy-cscore
  Building wheel for robotpy-cscore (PEP 517) ... error
  Complete output from command /home/pi/.virtualenvs/cv/bin/python3 /home/pi/.virtualenvs/cv/lib/python3.5/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmpob3vq7c0:
  ['cscore_src/cscore/src/main/native/cpp/PropertyContainer.cpp', 'cscore_src/cscore/src/main/native/cpp/CvSinkImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/MjpegServerImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/cscore_cpp.cpp', 'cscore_src/cscore/src/main/native/cpp/SourceImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/PropertyImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/cscore_oo.cpp', 'cscore_src/cscore/src/main/native/cpp/CvSourceImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/HttpCameraImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/UsbCameraImplCommon.cpp', 'cscore_src/cscore/src/main/native/cpp/SinkImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/cscore_c.cpp', 'cscore_src/cscore/src/main/native/cpp/Telemetry.cpp', 'cscore_src/cscore/src/main/native/cpp/Frame.cpp', 'cscore_src/cscore/src/main/native/cpp/Notifier.cpp', 'cscore_src/cscore/src/main/native/cpp/JpegUtil.cpp', 'cscore_src/cscore/src/main/native/cpp/Instance.cpp', 'cscore_src/cscore/src/main/native/linux/UsbCameraProperty.cpp', 'cscore_src/cscore/src/main/native/linux/UsbUtil.cpp', 'cscore_src/cscore/src/main/native/linux/UsbCameraImpl.cpp', 'cscore_src/cscore/src/main/native/linux/NetworkUtil.cpp', 'cscore_src/cscore/src/main/native/linux/NetworkListener.cpp']
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.5
  creating build/lib.linux-armv7l-3.5/cscore
  copying cscore/__main__.py -> build/lib.linux-armv7l-3.5/cscore
  copying cscore/cameraserver.py -> build/lib.linux-armv7l-3.5/cscore
  copying cscore/grip.py -> build/lib.linux-armv7l-3.5/cscore
  copying cscore/_info.py -> build/lib.linux-armv7l-3.5/cscore
  copying cscore/version.py -> build/lib.linux-armv7l-3.5/cscore
  copying cscore/__init__.py -> build/lib.linux-armv7l-3.5/cscore
  copying cscore/imagewriter.py -> build/lib.linux-armv7l-3.5/cscore
  copying cscore/_logging.py -> build/lib.linux-armv7l-3.5/cscore
  running build_ext
  creating tmp
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -I/home/pi/.virtualenvs/cv/include/python3.5m -c /tmp/tmp6a2epffo.cpp -o tmp/tmp6a2epffo.o -std=c++14
  cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -I/home/pi/.virtualenvs/cv/include/python3.5m -c /tmp/tmpcgj5m7lb.cpp -o tmp/tmpcgj5m7lb.o -fvisibility=hidden
  cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
  building 'cscore._cscore' extension
  Traceback (most recent call last):
    File "/home/pi/.virtualenvs/cv/lib/python3.5/site-packages/pip/_vendor/pep517/_in_process.py", line 207, in <module>
      main()
    File "/home/pi/.virtualenvs/cv/lib/python3.5/site-packages/pip/_vendor/pep517/_in_process.py", line 197, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/home/pi/.virtualenvs/cv/lib/python3.5/site-packages/pip/_vendor/pep517/_in_process.py", line 141, in build_wheel
      metadata_directory)
    File "/tmp/pip-build-env-lvo77bf1/overlay/lib/python3.5/site-packages/setuptools/build_meta.py", line 173, in build_wheel
      self.run_setup()
    File "/tmp/pip-build-env-lvo77bf1/overlay/lib/python3.5/site-packages/setuptools/build_meta.py", line 126, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 254, in <module>
      entry_points={"robotpylib": ["info = cscore._info:Info"]},
    File "/tmp/pip-build-env-lvo77bf1/overlay/lib/python3.5/site-packages/setuptools/__init__.py", line 145, in setup
[... cut ...]
      pp_opts = gen_preprocess_options(macros, incdirs)
    File "/usr/lib/python3.5/distutils/ccompiler.py", line 1075, in gen_preprocess_options
      pp_opts.append("-I%s" % dir)
    File "setup.py", line 66, in __str__
      import numpy as np
  ImportError: No module named 'numpy'

  ----------------------------------------
  Failed building wheel for robotpy-cscore
  Running setup.py clean for robotpy-cscore
Failed to build robotpy-cscore
Could not build wheels for {} which use PEP 517 and cannot be installed directly
(cv) pi@OpenCV:~ $

virtuald, 2019-02-13:
Looks like numpy isn't installed? Try this:

```
pip3 install robotpy-cscore --no-build-isolation
```

PepperProgramming, 2019-02-13:
numpy is installed 

From build o/p "Requirement already satisfied: numpy in ./.virtualenvs/cv/lib/python3.5/site-packages (from robotpy-cscore) (1.16.0). "

Here is my pip list
(cv) pi@OpenCV:~ $ pip list
Package                 Version
----------------------- ----------
Click                   7.0
cloudpickle             0.6.1
cycler                  0.10.0
dask                    1.0.0
decorator               4.3.0
dlib                    19.16.0
face-recognition        1.2.3
face-recognition-models 0.3.0
imutils                 0.5.2
kiwisolver              1.0.1
matplotlib              3.0.2
netifaces               0.10.9
networkx                2.2
numpy                   1.16.0
opencv-contrib-python   3.4.4.19
picamera                1.13
Pillow                  5.4.1
pip                     19.0.1
pynetworktables         2019.0.0
pyparsing               2.3.1
python-dateutil         2.7.5
PyWavelets              1.0.1
robotpy-cscore          2019.0.0a1
scikit-image            0.14.1
scipy                   1.2.0
setuptools              40.6.3
six                     1.12.0
toolz                   0.9.0
wheel                   0.32.3
(cv) pi@OpenCV:~ $

and when I tried pip3 install robotpy-cscore==2019.0.0a01, it worked without an error. 

Let me try pip3 install robotpy-cscore --no-build-isolation

PepperProgramming, 2019-02-13:
Thanks a lot, that worked, I see  robotpy-cscore 2019.0.7 installed.

closed 2019-02-13 by PepperProgramming