robotpy/robotpy-cscore #71: Installation errors

lu0079, 2019-03-21:
While trying to install cscore, I get this message
```
sudo pip3 install robotpy-cscore
Collecting robotpy-cscore
  Downloading https://files.pythonhosted.org/packages/16/42/7f1a58d74d583f05fc02b200d91036be6e5f8ce690f3d25136ccddb84761/robotpy-cscore-2019.1.0.tar.gz (912kB)
    100% |████████████████████████████████| 921kB 446kB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  Complete output from command /usr/bin/python3 /usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpufdfwt8x:
  Traceback (most recent call last):
    File "/usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py", line 207, in <module>
      main()
    File "/usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py", line 197, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py", line 48, in get_requires_for_build_wheel
      backend = _build_backend()
    File "/usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py", line 39, in _build_backend
      obj = getattr(obj, path_part)
  AttributeError: module 'setuptools.build_meta' has no attribute '__legacy__'
  
  ----------------------------------------
Command "/usr/bin/python3 /usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpufdfwt8x" failed with error code 1 in /tmp/pip-install-79h0qnb0/robotpy-cscore
```

virtuald, 2019-03-24:
That sounds like a setuptools error. Recommend you upgrade setuptools and pip. If that doesn't address it, can you paste the output of `pip list` here?

lu0079, 2019-03-27:
Package                       Version               
----------------------------- ----------------------
apturl                        0.5.2                 
atomicwrites                  1.3.0                 
attrs                         19.1.0                
beautifulsoup4                4.4.1                 
blinker                       1.3                   
Brlapi                        0.6.4                 
chardet                       2.3.0                 
checkbox-support              0.22                  
command-not-found             0.3                   
cryptography                  1.2.3                 
defer                         1.0.6                 
feedparser                    5.1.3                 
guacamole                     0.9.2                 
html5lib                      0.999                 
httplib2                      0.9.1                 
idna                          2.0                   
Jinja2                        2.8                   
language-selector             0.1                   
louis                         2.6.4                 
lxml                          3.5.0                 
Mako                          1.0.3                 
MarkupSafe                    0.23                  
more-itertools                6.0.0                 
numpy                         1.16.2                
oauthlib                      1.0.3                 
onboard                       1.2.0                 
opencv-python                 4.0.0.21              
padme                         1.1.1                 
pathlib2                      2.3.3                 
pexpect                       4.0.1                 
Pillow                        3.1.2                 
Pint                          0.9                   
pip                           19.0.3                
plainbox                      0.25                  
pluggy                        0.9.0                 
ptyprocess                    0.5                   
py                            1.8.0                 
pyasn1                        0.1.9                 
pybind11                      2.2.2                 
pycups                        1.9.73                
pycurl                        7.43.0                
pyfrc                         2019.1.0              
pygobject                     3.20.0                
PyJWT                         1.3.0                 
pynetconsole                  2.0.2                 
pynetworktables               2019.0.2              
pyparsing                     2.0.3                 
pyserial                      3.4                   
pytest                        4.3.0                 
python-apt                    1.1.0b1+ubuntu0.16.4.2
python-debian                 0.1.27                
python-systemd                231                   
pyxdg                         0.25                  
reportlab                     3.3.0                 
requests                      2.9.1                 
robotpy-ctre                  2019.3.2              
robotpy-hal-base              2019.2.1              
robotpy-hal-sim               2019.2.1              
robotpy-installer             2019.2.0              
robotpy-pathfinder            0.2.5                 
robotpy-wpilib-utilities      2019.0.7              
sessioninstaller              0.0.0                 
setuptools                    40.8.0                
six                           1.12.0                
system-service                0.3                   
ubuntu-drivers-common         0.0.0                 
ufw                           0.35                  
unattended-upgrades           0.1                   
unity-scope-calculator        0.1                   
unity-scope-chromiumbookmarks 0.1                   
unity-scope-colourlovers      0.1                   
unity-scope-devhelp           0.1                   
unity-scope-firefoxbookmarks  0.1                   
unity-scope-gdrive            0.7                   
unity-scope-manpages          0.1                   
unity-scope-openclipart       0.1                   
unity-scope-texdoc            0.1                   
unity-scope-tomboy            0.1                   
unity-scope-virtualbox        0.1                   
unity-scope-yelp              0.1                   
unity-scope-zotero            0.1                   
urllib3                       1.13.1                
usb-creator                   0.3.0                 
wheel                         0.33.0                
wpilib                        2019.2.1              
xdiagnose                     3.8.4.1               
xkit                          0.0.0                 
XlsxWriter                    0.7.3

auscompgeek, 2019-03-28:
Hmm... can't see anything obviously wrong. What about `pip list -v`?

lu0079, 2019-03-28:
<details>
Package                       Version                Location                                       Installer
----------------------------- ---------------------- ---------------------------------------------- ---------
apturl                        0.5.2                  /usr/lib/python3/dist-packages                          
atomicwrites                  1.3.0                  /home/linus/.local/lib/python3.5/site-packages pip      
attrs                         19.1.0                 /home/linus/.local/lib/python3.5/site-packages pip      
beautifulsoup4                4.4.1                  /usr/lib/python3/dist-packages                          
blinker                       1.3                    /usr/lib/python3/dist-packages                          
Brlapi                        0.6.4                  /usr/lib/python3/dist-packages                          
chardet                       2.3.0                  /usr/lib/python3/dist-packages                          
checkbox-support              0.22                   /usr/lib/python3/dist-packages                          
command-not-found             0.3                    /usr/lib/python3/dist-packages                          
cryptography                  1.2.3                  /usr/lib/python3/dist-packages                          
defer                         1.0.6                  /usr/lib/python3/dist-packages                          
feedparser                    5.1.3                  /usr/lib/python3/dist-packages                          
guacamole                     0.9.2                  /usr/lib/python3/dist-packages                          
html5lib                      0.999                  /usr/lib/python3/dist-packages                          
httplib2                      0.9.1                  /usr/lib/python3/dist-packages                          
idna                          2.0                    /usr/lib/python3/dist-packages                          
Jinja2                        2.8                    /usr/lib/python3/dist-packages                          
language-selector             0.1                    /usr/lib/python3/dist-packages                          
louis                         2.6.4                  /usr/lib/python3/dist-packages                          
lxml                          3.5.0                  /usr/lib/python3/dist-packages                          
Mako                          1.0.3                  /usr/lib/python3/dist-packages                          
MarkupSafe                    0.23                   /usr/lib/python3/dist-packages                          
more-itertools                6.0.0                  /home/linus/.local/lib/python3.5/site-packages pip      
numpy                         1.16.2                 /home/linus/.local/lib/python3.5/site-packages pip      
oauthlib                      1.0.3                  /usr/lib/python3/dist-packages                          
onboard                       1.2.0                  /usr/lib/python3/dist-packages                          
opencv-python                 4.0.0.21               /home/linus/.local/lib/python3.5/site-packages pip      
padme                         1.1.1                  /usr/lib/python3/dist-packages                          
pathlib2                      2.3.3                  /home/linus/.local/lib/python3.5/site-packages pip      
pexpect                       4.0.1                  /usr/lib/python3/dist-packages                          
Pillow                        3.1.2                  /usr/lib/python3/dist-packages                          
Pint                          0.9                    /home/linus/.local/lib/python3.5/site-packages pip      
pip                           19.0.3                 /usr/local/lib/python3.5/dist-packages         pip      
plainbox                      0.25                   /usr/lib/python3/dist-packages                          
pluggy                        0.9.0                  /home/linus/.local/lib/python3.5/site-packages pip      
ptyprocess                    0.5                    /usr/lib/python3/dist-packages                          
py                            1.8.0                  /home/linus/.local/lib/python3.5/site-packages pip      
pyasn1                        0.1.9                  /usr/lib/python3/dist-packages                          
pybind11                      2.2.2                  /usr/local/lib/python3.5/dist-packages         pip      
pycups                        1.9.73                 /usr/lib/python3/dist-packages                          
pycurl                        7.43.0                 /usr/lib/python3/dist-packages                          
pyfrc                         2019.1.0               /home/linus/.local/lib/python3.5/site-packages pip      
pygobject                     3.20.0                 /usr/lib/python3/dist-packages                          
PyJWT                         1.3.0                  /usr/lib/python3/dist-packages                          
pynetconsole                  2.0.2                  /home/linus/.local/lib/python3.5/site-packages pip      
pynetworktables               2019.0.2               /home/linus/.local/lib/python3.5/site-packages pip      
pyparsing                     2.0.3                  /usr/lib/python3/dist-packages                          
[... cut ...]
wheel                         0.33.0                 /home/linus/.local/lib/python3.5/site-packages pip      
wpilib                        2019.2.1               /home/linus/.local/lib/python3.5/site-packages pip      
xdiagnose                     3.8.4.1                /usr/lib/python3/dist-packages                          
xkit                          0.0.0                  /usr/lib/python3/dist-packages                          
XlsxWriter                    0.7.3                  /usr/lib/python3/dist-packages  

</details>

auscompgeek, 2019-03-28:
Looking at that output, I'm convinced you probably have mismatching versions of pip and setuptools at the system level. I can't say for certain though, since setuptools in that list is installed in your user, rather than from the system.

I also note that you have robotpy-ctre and robotpy-pathfinder installed at the system level, but other parts of RobotPy installed in your user site-packages.

Protips: never run pip as root, and don't try to use pip to manage packages installed by the OS package manager.

Good luck.

lu0079, 2019-03-28:
pip3 install --upgrade pip setuptools
Requirement already up-to-date: pip in /usr/local/lib/python3.5/dist-packages (19.0.3)
Requirement already up-to-date: setuptools in /home/linus/.local/lib/python3.5/site-packages (40.8.0)

It tells me its all up to date? Unless I'm missing something? 
Is there any other likely potential causes?

virtuald, 2019-03-28:
You could try downgrading to an older version of pip. It feels like there's something else going on here though.

Also I note that you have opencv-python installed, and unfortunately cscore won't work with that (see #4). While this (probably?) isn't the cause of your issue, once you figure this out you'll run into that.

Have you tried using the `--no-build-isolation` flag to pip install? The installation instructions recommend this.

lu0079, 2019-03-29:
Resolved through setting up virtual environment and affixing the --no-build-isolation flag. Thanks! :D

closed 2019-03-29 by lu0079

auscompgeek, 2019-03-29:
Awesome. Your system is probably still broken though.