robotpy/robotpy-cscore #72: getting invalid jpeg on ubuntu 18

oriagranat9, 2019-03-30:
while using the quick camera server example on ubuntu 18 im getting a warning:
WARNING:cscore:USB Camera 0: invalid JPEG image received from camera

the server opens and i can browse to the website but i dont get any image

auscompgeek, 2019-03-30:
It is likely that the camera you are using does not support the MJPEG format, which CSCore uses by default. Please check the video modes the camera supports (which are listed at the bottom of the webpage of the camera server).

oriagranat9, 2019-03-30:
Is there any way to use other formats with cscore

auscompgeek, 2019-03-30:
- https://robotpy.readthedocs.io/projects/cscore/en/stable/objects.html#cscore.VideoSource.setPixelFormat
- https://robotpy.readthedocs.io/projects/cscore/en/stable/objects.html#cscore.VideoSource.setVideoMode

virtuald, 2019-04-01:
What kind of camera is it?

prensing, 2019-12-26:
My guess, given oriagranat9 other bugs is this is a PS Eye camera. AFAIK, it does not support MJPEG, only YUYV. 

You can select YUYV with
        # PS Eye camera needs to have its pixelformat set
        camera.setPixelFormat(cscore.VideoMode.PixelFormat.kYUYV)

auscompgeek, 2019-12-27:
I'm tempted to call this an upstream documentation bug.