robotpy/robotpy-cscore #74: Unknown symbol on import

132ikl, 2019-04-08:
System: aarch64
```
>>> import cscore
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tam/ColorTrack/venv/lib/python3.6/site-packages/cscore/__init__.py", line 1, in <module>
    from ._cscore import *
ImportError: /home/tam/ColorTrack/venv/lib/python3.6/site-packages/cscore/_cscore.cpython-36m-aarch64-linux-gnu.so: undefined symbol: _ZN2cv3Mat20updateContinuityFlagEv
```

After doing `pip3 install --no-build-isolation --no-cache-dir robotpy-cscore` in a fresh virtualenv

virtuald, 2019-04-08:
Sounds like it's linking to the wrong version of opencv. What's the output of `ldd /home/tam/ColorTrack/venv/lib/python3.6/site-packages/cscore/_cscore.cpython-36m-aarch64-linux-gnu.so`?

132ikl, 2019-04-08:
```ldd /home/tam/ColorTrack/venv/lib/python3.6/site-packages/cscore/_cscore.cpython-36m-aarch64-linux-gnu.so
        linux-vdso.so.1 (0x0000007f808cc000)
        libopencv_core.so.3.3 => /usr/lib/libopencv_core.so.3.3 (0x0000007f803b6000)
        libopencv_imgproc.so.3.3 => /usr/lib/libopencv_imgproc.so.3.3 (0x0000007f7fbe9000)
        libopencv_imgcodecs.so.3.3 => /usr/lib/libopencv_imgcodecs.so.3.3 (0x0000007f7fac5000)
        libstdc++.so.6 => /usr/lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f7f930000)
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f7f876000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f7f852000)
        libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f7f826000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f7f6cd000)
        /lib/ld-linux-aarch64.so.1 (0x0000007f808a1000)
        libtbb.so.2 => /usr/lib/aarch64-linux-gnu/libtbb.so.2 (0x0000007f7f68f000)
        libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007f7f662000)
        libjpeg.so.8 => /usr/lib/aarch64-linux-gnu/libjpeg.so.8 (0x0000007f7f618000)
        libpng16.so.16 => /usr/lib/aarch64-linux-gnu/libpng16.so.16 (0x0000007f7f5dd000)
        libtiff.so.5 => /usr/lib/aarch64-linux-gnu/libtiff.so.5 (0x0000007f7f562000)
        libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f7f54d000)
        liblzma.so.5 => /lib/aarch64-linux-gnu/liblzma.so.5 (0x0000007f7f51d000)
        libjbig.so.0 => /usr/lib/aarch64-linux-gnu/libjbig.so.0 (0x0000007f7f500000)```

virtuald, 2019-04-08:
`libopencv_imgproc.so.3.3` et al indicates that it's trying to use OpenCV 3.3. Did you install 3.4.5 manually? 3.3 might be installed by the system then... I suggest removing 3.3 via your system package manager and that should fix your issue.

132ikl, 2019-04-08:
That looks to be the issue. Recompiling now.

132ikl, 2019-04-08:
It works now. Thank you so much!

closed 2019-04-08 by 132ikl