robotpy/robotpy-cscore #77: gettting nothing while running cam in process

oriagranat9, 2019-05-12:
while trying to run a usb cam in a process and grabbing its frames the value of the image is 0 
`
#camera is a usb camera object
multiprocessing.Process(target=self.camera_process, args=(cam,camera_server))

 def camera_process(self, camera,cs):
   cv_sink = cscore.CvSink("test")
   cv_sink.setSource(camera)
   while True:
            _, image = cv_sink.grabFrame(image)
`

PeterJohnson, 2019-05-12:
No, this won't work.  You're going to have to send the images to the subprocess yourself (e.g. via a pipe).

virtuald, 2019-05-12:
@oriagranat9 what are you trying to accomplish? If you're trying to run this code on a roborio, see the documentation at https://robotpy.readthedocs.io/en/stable/vision/roborio.html

oriagranat9, 2019-05-12:
no im trying to run this on my computer in and to process each camera in parallel

virtuald, 2019-05-12:
If you create your camera objects inside the `camera_process` function, it should work.

virtuald, 2020-02-09:
Closed due to age.

closed 2020-02-09 by virtuald