robotpy/robotpy-cscore #78: set exposure bug

oriagranat9, 2019-06-15:
while setting manual exposure on a `cscore._cscore.UsbCamera` object 
using `usb_cam_obj.setExposureManual(some int between 0 to 100)`
i get `CS: ERROR: ioctl VIDIOC_S_EXT_CTRLS failed at UsbCameraProperty.cpp:65: Invalid argument (UsbUtil.cpp:156)`

virtuald, 2019-06-15:
Sounds like a bug in the underlying cscore library. @PeterJohnson?

PeterJohnson, 2019-06-26:
It means the underlying V4L call failed.  This can happen for a variety of reasons... the property may not exist, or it's not a valid value to set.  What camera make/model is this?  Can you share the output of `v4l2-ctl --list-ctrls` for the camera device?

oriagranat9, 2019-06-26:
the camera is a playstation eye and the l4v2 logf is `v4l2-ctl -d /dev/video0 --list-ctrls

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=0 value=48 flags=slider
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=32 value=32 flags=slider
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=64 value=64 flags=slider
                            hue 0x00980903 (int)    : min=-90 max=90 step=1 default=0 value=0 flags=slider
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                       exposure 0x00980911 (int)    : min=0 max=255 step=1 default=120 value=120 flags=inactive, volatile
                 gain_automatic 0x00980912 (bool)   : default=1 value=1 flags=update
                           gain 0x00980913 (int)    : min=0 max=63 step=1 default=20 value=20 flags=inactive, volatile
                horizontal_flip 0x00980914 (bool)   : default=0 value=0
                  vertical_flip 0x00980915 (bool)   : default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=1 default=0 value=0
                      sharpness 0x0098091b (int)    : min=0 max=63 step=1 default=0 value=0 flags=slider

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=1 default=0 value=0 flags=update
`

PeterJohnson, 2019-07-01:
The problem is that the properties are called "auto_exposure" and "exposure" instead of what the library expects, which is "exposure_auto" and "exposure_absolute" respectively.  This will need a fix to the library to handle this case; what USB device and model ID is the camera? (lsusb I think can get this info).

As a workaround, you can use the string property methods instead to set the properties using the correct names.  E.g. something like `source.getProperty("auto_exposure").set(1)` followed by `source.getProperty("exposure").set(val * 100 / 255)`.

auscompgeek, 2019-07-02:
I have a PS Eye myself. Vendor and product ID are 1415:2000.

```console
$ lsusb
Bus 001 Device 035: ID 1415:2000 Nam Tai E&E Products Ltd. or OmniVision Technologies, Inc. Sony Playstation Eye
$ lsusb.py
 1-1             1415:2000 00  2.00  480MBit/s 500mA 3IFs (OmniVision Technologies, Inc. USB Camera-B4.09.24.1)
```

Ping me when you patch this and I'll verify it if I have the chance.

oriagranat9, 2019-08-11:
Any update on the matter?

auscompgeek, 2020-01-31:
So a quirk has been added to CSCore for 2020. We haven't cut a release yet but if you use robotpy-cscore master it should work.

auscompgeek, 2020-02-04:
We've released robotpy-cscore 2020.0.0.

closed 2020-02-04 by auscompgeek