robotpy/robotpy-cscore #87: Cannot compile cscore on raspberry 3 B opencv3.4.3

or-dan123, 2020-01-30:
Hello I am trying to compile the newest version of cscore for use on raspberry.
opencv 3.4.3 (compiled with enable shared library support)
I have done the following:
git clone --recursive https://github.com/robotpy/robotpy-cscore.git
python3 setup.py install

I have also installed via apt pkgconfig, setuptools and numpy.

when I get the following error:

```

['cscore_src/cscore/src/main/native/cpp/CvSourceImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/CvSinkImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/MjpegServerImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/cscore_oo.cpp', 'cscore_src/cscore/src/main/native/cpp/PropertyContainer.cpp', 'cscore_src/cscore/src/main/native/cpp/RawSourceImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/RawSinkImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/ConfigurableSourceImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/HttpCameraImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/Notifier.cpp', 'cscore_src/cscore/src/main/native/cpp/cscore_cpp.cpp', 'cscore_src/cscore/src/main/native/cpp/Instance.cpp', 'cscore_src/cscore/src/main/native/cpp/Frame.cpp', 'cscore_src/cscore/src/main/native/cpp/cscore_c.cpp', 'cscore_src/cscore/src/main/native/cpp/JpegUtil.cpp', 'cscore_src/cscore/src/main/native/cpp/UsbCameraImplCommon.cpp', 'cscore_src/cscore/src/main/native/cpp/SinkImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/PropertyImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/SourceImpl.cpp', 'cscore_src/cscore/src/main/native/cpp/Telemetry.cpp', 'cscore_src/cscore/src/main/native/linux/UsbCameraProperty.cpp', 'cscore_src/cscore/src/main/native/linux/NetworkUtil.cpp', 'cscore_src/cscore/src/main/native/linux/NetworkListener.cpp', 'cscore_src/cscore/src/main/native/linux/UsbUtil.cpp', 'cscore_src/cscore/src/main/native/linux/UsbCameraImpl.cpp']
running install
running bdist_egg
running egg_info
creating robotpy_cscore.egg-info
writing entry points to robotpy_cscore.egg-info/entry_points.txt
writing requirements to robotpy_cscore.egg-info/requires.txt
writing top-level names to robotpy_cscore.egg-info/top_level.txt
writing dependency_links to robotpy_cscore.egg-info/dependency_links.txt
writing robotpy_cscore.egg-info/PKG-INFO
writing manifest file 'robotpy_cscore.egg-info/SOURCES.txt'
reading manifest file 'robotpy_cscore.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no previously-included files found matching 'pyproject.toml'
writing manifest file 'robotpy_cscore.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-3.5
creating build/lib.linux-armv7l-3.5/cscore
copying cscore/grip.py -> build/lib.linux-armv7l-3.5/cscore
copying cscore/_info.py -> build/lib.linux-armv7l-3.5/cscore
copying cscore/version.py -> build/lib.linux-armv7l-3.5/cscore
copying cscore/__init__.py -> build/lib.linux-armv7l-3.5/cscore
copying cscore/imagewriter.py -> build/lib.linux-armv7l-3.5/cscore
copying cscore/__main__.py -> build/lib.linux-armv7l-3.5/cscore
copying cscore/_logging.py -> build/lib.linux-armv7l-3.5/cscore
copying cscore/cameraserver.py -> build/lib.linux-armv7l-3.5/cscore
running build_ext
creating tmp
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c /tmp/tmp4_es0fwr.cpp -o tmp/tmp4_es0fwr.o -std=c++17
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c /tmp/tmp5p0f5zfa.cpp -o tmp/tmp5p0f5zfa.o -fvisibility=hidden
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
building '_cscore' extension
creating build/temp.linux-armv7l-3.5
creating build/temp.linux-armv7l-3.5/src
creating build/temp.linux-armv7l-3.5/cscore_src
creating build/temp.linux-armv7l-3.5/cscore_src/cscore
creating build/temp.linux-armv7l-3.5/cscore_src/cscore/src
creating build/temp.linux-armv7l-3.5/cscore_src/cscore/src/main
creating build/temp.linux-armv7l-3.5/cscore_src/cscore/src/main/native
creating build/temp.linux-armv7l-3.5/cscore_src/cscore/src/main/native/cpp
creating build/temp.linux-armv7l-3.5/cscore_src/cscore/src/main/native/linux
creating build/temp.linux-armv7l-3.5/cscore_src/wpiutil
creating build/temp.linux-armv7l-3.5/cscore_src/wpiutil/src
creating build/temp.linux-armv7l-3.5/cscore_src/wpiutil/src/main
creating build/temp.linux-armv7l-3.5/cscore_src/wpiutil/src/main/native
creating build/temp.linux-armv7l-3.5/cscore_src/wpiutil/src/main/native/cpp
creating build/temp.linux-armv7l-3.5/cscore_src/wpiutil/src/main/native/cpp/uv
creating build/temp.linux-armv7l-3.5/cscore_src/wpiutil/src/main/native/cpp/llvm
creating build/temp.linux-armv7l-3.5/cscore_src/wpiutil/src/main/native/libuv
creating build/temp.linux-armv7l-3.5/cscore_src/wpiutil/src/main/native/libuv/src
creating build/temp.linux-armv7l-3.5/cscore_src/wpiutil/src/main/native/libuv/src/unix
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ipybind11/include -Icscore_src/cscore/src/main/native/include -Icscore_src/cscore/src/main/native/cpp -Icscore_src/wpiutil/src/main/native/include -Icscore_src/wpiutil/src/main/native/libuv/src -Icscore_src/wpiutil/src/main/native/libuv/include -I/usr/local/lib/python3.5/dist-packages/numpy/core/include -I/usr/local/include/opencv -I/usr/local/include -I/usr/include/python3.5m -c src/_cscore.cpp -o build/temp.linux-armv7l-3.5/src/_cscore.o -DVERSION_INFO="2019.1.0.post0.dev24" -s -g0 -std=c++17 -fvisibility=hidden -D_GNU_SOURCE
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
In file included from cscore_src/wpiutil/src/main/native/include/wpi/ArrayRef.h:15:0,
                 from cscore_src/cscore/src/main/native/include/cscore_cpp.h:17,
                 from cscore_src/cscore/src/main/native/include/cscore_oo.h:16,
                 from src/_cscore.cpp:8:
cscore_src/wpiutil/src/main/native/include/wpi/STLExtras.h:585:13: error: ‘optional’ in namespace ‘std’ does not name a template type
     -> std::optional<typename std::remove_const<
             ^~~~~~~~
cscore_src/wpiutil/src/main/native/include/wpi/STLExtras.h:585:21: error: expected initializer before ‘<’ token
     -> std::optional<typename std::remove_const<
                     ^
cscore_src/wpiutil/src/main/native/include/wpi/STLExtras.h:594:12: error: ‘optional’ in namespace ‘std’ does not name a template type
       std::optional<typename std::remove_const<typename std::remove_reference<
            ^~~~~~~~
In file included from cscore_src/cscore/src/main/native/include/cscore_cpp.h:17:0,
                 from cscore_src/cscore/src/main/native/include/cscore_oo.h:16,
                 from src/_cscore.cpp:8:
cscore_src/wpiutil/src/main/native/include/wpi/ArrayRef.h:65:44: error: invalid use of ‘::’
     /*implicit*/ ArrayRef(std::nullopt_t) {}
                                            ^
cscore_src/wpiutil/src/main/native/include/wpi/ArrayRef.h:65:44: error: expected ‘;’ at end of member declaration
cscore_src/wpiutil/src/main/native/include/wpi/ArrayRef.h:296:48: error: function definition does not declare parameters
     /*implicit*/ MutableArrayRef(std::nullopt_t) : ArrayRef<T>() {}
                                                ^
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

```

virtuald, 2020-01-30:
Sounds like you're using a version of GCC that doesn't fully support C++17. What version of gcc?

or-dan123, 2020-01-30:
Strange I have accidentally grabbed an antique image of raspbian 9 with an old gcc version.

#gcc -v                                                                                                                                                                                                                                                          Using built-in specs.                                                                                                                                                                                                                                                          COLLECT_GCC=gcc                                                                                                                                                                                                                                                                COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper                                                                                                                                                                                                             Target: arm-linux-gnueabihf                                                                                                                                                                                                                                                    Configured with: ../src/configure -v --with-pkgversion='Raspbian 6.3.0-18+rpi1+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf                                                                                                                              Thread model: posix                                                                                                                                                                                                                                                            gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1)

I will attempt to compile on a newer version

auscompgeek, 2020-01-30:
I would try building on Raspbian 10 (buster) first. Given Debian 9 (stretch) was released in 2017, I'm not surprised its gcc doesn't fully support C++17.

or-dan123, 2020-01-30:
Yes I made an buster image and a stretch image, and I by mistake took wrong one.

or-dan123, 2020-01-30:
I compiled it no problem on a newer installation!

however I am interested the version I compiled is for 2019 which isn't compatible with the playstation eye camera.
I have heard that there is a 2020 version which does support it, where can I obtain the 2020 version source?

virtuald, 2020-01-31:
Despite the 2019 label, if you cloned from master that is the 2020 code.

closed 2020-01-31 by virtuald