robotpy/robotpy-cscore #105: Program freezes when using "addAxisCamera"

harrison6462, 2021-03-10:
I'm using the sample code to use cscore to create a daemon process to run my vision code in python. When I try to use cscore.addAxisCamera, a large amount of the time the python program starts up and gets stuck inside the method permanently, but sometimes it works (the outcome was nondeterministic). We couldn't figure out anything that differs between the runs where it works vs. those where it freezes, and there's no error logging since it never exits the method.

virtuald, 2021-03-11:
What platform?

harrison6462, 2021-03-11:
Sorry I'm not sure what platform you're referring to. We're running it on the most up to date RoboRIO with the standard robotpy + robotpy-cscore libraries.
Also, I should probably mention that using the Java implementation of addAxisCamera always works, so it's not a matter of the hardware.

virtuald, 2021-03-12:
Do you have the ability to install gdb and get a stack trace of when it's frozen? That could be helpful. Another thing that could be helpful is to have a minimal example that reproduces it? I just tried to reproduce it and wasn't able to -- but I guess that makes sense since you said it wasn't deterministic.

Looking at it, I think it's likely that there's a GIL related deadlock? It's a pretty simple change to fix it, but I don't have time to fix it tonight *and* build a new wheel. Will try to get to it tomorrow night and see if that fixes it for you.

harrison6462, 2021-03-12:
Minimal example:
cs = CameraServer.getInstance()
print('entering addAxiscamera')
cs.addAxisCamera('169.254.241.45', name='axis-cam')
print('I suceeded!')

I deployed this 10 times and it worked on the first, fourth, ninth, and tenth times and froze the rest (we're running the python process with unbuffered printing to stdout/stderr).

I've been trying to get gdb installed on my RIO but I can't figure out how to install it without debian / apt-get.

virtuald, 2021-03-12:
I was going to tell you to use robotpy-installer to download/install it, but it seems that it doesn't work (reference: https://github.com/robotpy/roborio-packages/issues/29). 

I'll take a look at it tonight and see if I can duplicate with that snippet -- thats seems like a pretty easy thing to try.

virtuald, 2021-03-13:
I was able to duplicate the issue with that snippet, and I'm building a new release. robotpy-cscore 2021.0.2 will have the fix in it.

commit 40b78749 (2021-03-13): Release the GIL in various constructors

- Was causing a deadlock on the RoboRIO
- Fixes #105
closed 2021-03-13 by virtuald