robotpy/robotpy-cscore #106: Installing in PyCharm Virtual Enviornment

jonathandao0, 2021-10-30:
I'm trying to install robotpy-cscore in PyCharm on Windows with opencv-python v4.4.0.46. Is this currently possible? If so, what additional steps do I need to do?

I've tried installing via robotpy_installer with the following command:
`robotpy-installer download robotpy-cscore`

This seems to install successfully, but PyCharm doesn't recognize cscore as a valid import if installed this way

Installing via the command prompt with pip gives the following error:
```
python -m pip install robotpy-cscore

`Collecting robotpy-cscore
  Using cached robotpy-cscore-2021.0.2.tar.gz (993 kB)
  Preparing metadata (setup.py) ... done
Collecting numpy
  Using cached numpy-1.21.3-cp37-cp37m-win_amd64.whl (14.0 MB)
Collecting pynetworktables
  Using cached pynetworktables-2021.0.0-py3-none-any.whl (65 kB)
Building wheels for collected packages: robotpy-cscore
  Building wheel for robotpy-cscore (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\Jonathan\AppData\Local\Programs\Python\Python37\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Jonathan\\AppData
\\Local\\Temp\\pip-install-fwvg89s9\\robotpy-cscore_006f62a6394d43b3a7c21141fb9226a3\\setup.py'"'"'; __file__='"'"'C:\\Users\\Jonathan\\AppData\\Local\\Temp\\pip-install-fwvg89s9
\\robotpy-cscore_006f62a6394d43b3a7c21141fb9226a3\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setu
ptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\Jonatha
n\AppData\Local\Temp\pip-wheel-kcyz2njz'
       cwd: C:\Users\Jonathan\AppData\Local\Temp\pip-install-fwvg89s9\robotpy-cscore_006f62a6394d43b3a7c21141fb9226a3\
  Complete output (46 lines):
  WARNING: pkg-config probably not installed: FileNotFoundError(2, 'The system cannot find the file specified', None, 2, None) - continuing anyway
  ['cscore_src/cscore/src/main/native/cpp\\ConfigurableSourceImpl.cpp', 'cscore_src/cscore/src/main/native/cpp\\cscore_c.cpp', 'cscore_src/cscore/src/main/native/cpp\\cscore_cpp.
cpp', 'cscore_src/cscore/src/main/native/cpp\\cscore_oo.cpp', 'cscore_src/cscore/src/main/native/cpp\\CvSinkImpl.cpp', 'cscore_src/cscore/src/main/native/cpp\\CvSourceImpl.cpp',
'cscore_src/cscore/src/main/native/cpp\\Frame.cpp', 'cscore_src/cscore/src/main/native/cpp\\HttpCameraImpl.cpp', 'cscore_src/cscore/src/main/native/cpp\\Instance.cpp', 'cscore_sr
c/cscore/src/main/native/cpp\\JpegUtil.cpp', 'cscore_src/cscore/src/main/native/cpp\\MjpegServerImpl.cpp', 'cscore_src/cscore/src/main/native/cpp\\Notifier.cpp', 'cscore_src/csco
re/src/main/native/cpp\\PropertyContainer.cpp', 'cscore_src/cscore/src/main/native/cpp\\PropertyImpl.cpp', 'cscore_src/cscore/src/main/native/cpp\\RawSinkImpl.cpp', 'cscore_src/c
score/src/main/native/cpp\\RawSourceImpl.cpp', 'cscore_src/cscore/src/main/native/cpp\\SinkImpl.cpp', 'cscore_src/cscore/src/main/native/cpp\\SourceImpl.cpp', 'cscore_src/cscore/
src/main/native/cpp\\Telemetry.cpp', 'cscore_src/cscore/src/main/native/cpp\\UsbCameraImplCommon.cpp', 'cscore_src/cscore/src/main/native/windows\\COMCreators.cpp', 'cscore_src/c
score/src/main/native/windows\\NetworkListener.cpp', 'cscore_src/cscore/src/main/native/windows\\NetworkUtil.cpp', 'cscore_src/cscore/src/main/native/windows\\UsbCameraImpl.cpp',
 'cscore_src/cscore/src/main/native/windows\\UsbCameraProperty.cpp', 'cscore_src/cscore/src/main/native/windows\\WindowsMessagePump.cpp']
  C:\Users\Jonathan\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\_distributor_init.py:32: UserWarning: loaded more than 1 DLL from .libs:
  C:\Users\Jonathan\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\.libs\libopenblas.PYQHXLVVQ7VESDPUVUADXEVJOBGHJPAY.gfortran-win_amd64.dll
  C:\Users\Jonathan\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\.libs\libopenblas.XWYDX2IKJW2NMTWSFYNGFUWKQU3LYTCZ.gfortran-win_amd64.dll
    stacklevel=1)
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.7
  creating build\lib.win-amd64-3.7\cscore
  copying cscore\cameraserver.py -> build\lib.win-amd64-3.7\cscore
  copying cscore\grip.py -> build\lib.win-amd64-3.7\cscore
  copying cscore\imagewriter.py -> build\lib.win-amd64-3.7\cscore
  copying cscore\version.py -> build\lib.win-amd64-3.7\cscore
  copying cscore\_info.py -> build\lib.win-amd64-3.7\cscore
  copying cscore\_logging.py -> build\lib.win-amd64-3.7\cscore
  copying cscore\__init__.py -> build\lib.win-amd64-3.7\cscore
  copying cscore\__main__.py -> build\lib.win-amd64-3.7\cscore
  running build_ext
  building '_cscore' extension
  creating build\temp.win-amd64-3.7
  creating build\temp.win-amd64-3.7\Release
  creating build\temp.win-amd64-3.7\Release\src
  creating build\temp.win-amd64-3.7\Release\cscore_src
  creating build\temp.win-amd64-3.7\Release\cscore_src\cscore
  creating build\temp.win-amd64-3.7\Release\cscore_src\cscore\src
  creating build\temp.win-amd64-3.7\Release\cscore_src\cscore\src\main
  creating build\temp.win-amd64-3.7\Release\cscore_src\cscore\src\main\native
  creating build\temp.win-amd64-3.7\Release\cscore_src\cscore\src\main\native\cpp
  creating build\temp.win-amd64-3.7\Release\cscore_src\cscore\src\main\native\windows
  creating build\temp.win-amd64-3.7\Release\cscore_src\wpiutil
  creating build\temp.win-amd64-3.7\Release\cscore_src\wpiutil\src
[... 78 lines cut ...]
    creating build\temp.win-amd64-3.7\Release\cscore_src\wpiutil\src\main\native\cpp
    creating build\temp.win-amd64-3.7\Release\cscore_src\wpiutil\src\main\native\cpp\llvm
    creating build\temp.win-amd64-3.7\Release\cscore_src\wpiutil\src\main\native\cpp\uv
    creating build\temp.win-amd64-3.7\Release\cscore_src\wpiutil\src\main\native\libuv
    creating build\temp.win-amd64-3.7\Release\cscore_src\wpiutil\src\main\native\libuv\src
    creating build\temp.win-amd64-3.7\Release\cscore_src\wpiutil\src\main\native\libuv\src\win
    C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ipybind11/include -Icscore
_src/cscore/src/main/native/include -Icscore_src/cscore/src/main/native/cpp -Icscore_src/wpiutil/src/main/native/include -Icscore_src/wpiutil/src/main/native/libuv/src -Icscore_s
rc/wpiutil/src/main/native/libuv/include -IC:\Users\Jonathan\AppData\Local\Programs\Python\Python37\lib\site-packages\numpy\core\include -IC:\Users\Jonathan\AppData\Local\Program
s\Python\Python37\include -IC:\Users\Jonathan\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.
30037\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\incl
ude\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Wind
ows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt
" /EHsc /Tpsrc/_cscore.cpp /Fobuild\temp.win-amd64-3.7\Release\src/_cscore.obj /EHsc /DNOMINMAX /std:c++17 /DVERSION_INFO=\"2021.0.2\"
    _cscore.cpp
    src/_cscore.cpp(12): fatal error C1083: Cannot open include file: 'opencv2/opencv.hpp': No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30037\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\Jonathan\AppData\Local\Programs\Python\Python37\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0]
= '"'"'C:\\Users\\Jonathan\\AppData\\Local\\Temp\\pip-install-fwvg89s9\\robotpy-cscore_006f62a6394d43b3a7c21141fb9226a3\\setup.py'"'"'; __file__='"'"'C:\\Users\\Jonathan\\AppData
\\Local\\Temp\\pip-install-fwvg89s9\\robotpy-cscore_006f62a6394d43b3a7c21141fb9226a3\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file
__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"')
)' install --record 'C:\Users\Jonathan\AppData\Local\Temp\pip-record-2l5okr_4\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\Jonath
an\AppData\Local\Programs\Python\Python37\Include\robotpy-cscore' Check the logs for full command output.
```

virtuald, 2021-10-30:
Not without a lot of work. See also:

* https://github.com/robotpy/robotpy-cscore/issues/4
* https://github.com/robotpy/robotpy-cscore/issues/17

I think @auscompgeek has a build in conda forge that works on Windows? Haven't really used it though.

closed 2021-10-30 by virtuald

auscompgeek, 2021-10-31:
Yeah, the conda-forge package is here: https://github.com/conda-forge/robotpy-cscore-feedstock