robotpy/robotpy-cscore #107: [BUG]: Server and CvSource constructors hang; cannot use CTRL-C

pinkbluesky, 2022-01-25:
### Problem description

When I run the code snippet to `addServer` to the `CameraServer` instance, the program sporadically hangs on that line. Then I have to terminate the Python script with `CTRL-\` because `CTRL-C` doesn't work. Then when I try it again, it usually makes it past that hang, but then hangs at the next line, where I set the `CvSource` for the server. Then I have to terminate again, run the script again, and usually on this third try the program will make it through. 

I haven't been able to discern a particular reason as to why it hangs, it just does sometimes but once I restart the script twice it's usually back to normal. 

### Operating System

Linux

### Reproducible example code

```text
from cscore import CameraServer, CvSource, VideoMode

# Initialize a camera server
cam_server = CameraServer.getInstance()

# Add a camera server
print('Attempting add a MjpegServer')
server = cam_server.addServer(name='0')
print('Completed attempt to add server')
stream = CvSource('0', VideoMode.PixelFormat.kMJPEG, 160, 120, 10)
server.setSource(stream)
print('CvSource has been set for server at port ' + str(server.getPort()))
```

virtuald, 2022-01-25:
Which version of cscore, which version of Python, and on what platform (x86, rpi, etc... )

auscompgeek, 2022-01-25:
Also: what Linux distro?

pinkbluesky, 2022-01-26:
Running `python3 --version` gives `Python 3.6.9`
Running `python --version` gives `Python 2.7.17`

I always run the Python code in question using `python3` in the terminal.

Running `pip3 list` shows `robotpy-cscore` version `2021.0.0`. `pip list` also shows the `2021` version.

Running `uname -m` gives `aarch64`. `uname -r` gives `4.9.140-tegra`.

```
NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

Also it's probably helpful to note that I'm running this code on a Nvidia Jetson Nano.
Thanks.

auscompgeek, 2022-01-26:
Ah, this is probably #105. I'm guessing you're using my distro packages from build.opensuse.org? Looks like I forgot to update the robotpy-cscore package there with that bugfix. I'll push up 2021.0.2 in a bit.

pinkbluesky, 2022-01-26:
Yep, I downloaded the debian file from opensuse here (https://software.opensuse.org/download.html?project=home%3Aauscompgeek%3Arobotpy&package=python3-cscore) for Ubuntu.

auscompgeek, 2022-01-26:
Updated the OBS repo. Your 2021.0.2 bionic aarch64 package should be available for download now. Let us know how it goes.

pinkbluesky, 2022-01-26:
The CameraServer code is no longer hanging, but I've also got an issue with the pynetworktables code hanging. It's on version `2021.0.0`, and it doesn't look like any updates are available through `pip3 install --upgrade pynetworktables`. Any idea why this could be happening?

Code snippet:
```
# Start thread to connect to NetworkTables
cond = threading.Condition()
notified = [False]

def connection_listener(connected, info):
    print(info, '; Connected=%s' % connected)
    with cond:
        notified[0] = True
        cond.notify()

# Use RoboRIO static IP address
NetworkTables.initialize(server='10.1.92.2')
NetworkTables.addConnectionListener(connection_listener, immediateNotify=True)

with cond:
    print('Waiting')
    if not notified[0]:
        cond.wait()

print('Connected to NetworkTables!')
```

I've been running this code without issues for the past couple of weeks, but it suddenly stopped working. The cscore update hasn't changed the status of this error.

virtuald, 2022-12-05:
Sorry we didn't get back to you last year! But it sounds like your cscore issue has been resolved. In 2023 everything will be switching over to ntcore, so I expect this will disappear.

closed 2022-12-05 by virtuald