robotpy/robotpy-cscore #113: Installing cscore on 32-bit Buster OS fails

vyang2968, 2022-03-15:
### Problem description

I am trying to install cscore on my Raspberry Pi 4, 32-Bit buster OS Lite. As far as I know, you can do this through `pip3 install robotpy-cscore` or following the instructions at [https://software.opensuse.org/download.htmlproject=home%3Aauscompgeek%3Arobotpy&package=python3-cscore]. I have tried the second one, but it says `E: Unable to locate package robotpy-cscore`. That's why I went onto installing from pip. With this approach, 
`src/_cscore.cpp:12:10: fatal error: opencv2/opencv.hpp: No such file or directory
     #include <opencv2/opencv.hpp>
              ^~~~~~~~~~~~~~~~~~~~
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1`

Any help with either of these would be much appreciated. 

### Operating System

Raspbian

virtuald, 2022-03-15:
Have you read the section about [compiling from source](https://robotpy.readthedocs.io/en/stable/install/cscore.html#compiling-from-source)? If you get that error, it means you don't have OpenCV installed.

vyang2968, 2022-03-15:
I've tried compiling from source before and it did not work for me. I've worked around this issue in the past by following the instructions on this [website](https://software.opensuse.org/download.html?project=home%3Aauscompgeek%3Arobotpy&package=python3-cscore) but that was on 64-bit Bullseye. Ideally I'd like to get the same process working with this operating system that I'm on.

virtuald, 2022-03-15:
@auscompgeek any chance you can update that package?

auscompgeek, 2022-03-15:
The OBS package does have [python3-cscore](https://software.opensuse.org/download.html?project=home%3Aauscompgeek%3Arobotpy&package=python3-cscore) binary packages for Raspbian 10 and 11. Unfortunately OBS download pages seem to not properly support Raspbian, so it doesn't provide install instructions. The packages are listed under `Unknown` however.

If you want to do an online install, you can use the Debian instructions, but replace `Debian` with `Raspbian` in the URLs.

vyang2968, 2022-03-21:
Thank you so much! It worked perfectly, I verified it with `python3 -c "import cscore"`.
For anyone wanting to just copy and paste, here's the scripts:

`echo 'deb http://download.opensuse.org/repositories/home:/auscompgeek:/robotpy/Raspbian_10/ /' | sudo tee /etc/apt/sources.list.d/home:auscompgeek:robotpy.list`

`curl -fsSL https://download.opensuse.org/repositories/home:auscompgeek:robotpy/Raspbian_10/Release.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/home_auscompgeek_robotpy.gpg > /dev/null`

`sudo apt update`

`sudo pip3 install robotpy-cscore`

closed 2022-03-21 by vyang2968